rolker/unh_marine_autonomy #4: Getting started - are errors expected?

bsb808, 2021-07-01:
I have been able to run the project11 demo

```
roslaunch project11_simulation sim_local.launch
```

The Qt window launches, but I do get a number of errors.  Before I dig into each, I wanted to check to see that these are errors that are not to be expected:

### pid
process[ben/dubins_curves-5]: started with pid [1931]
ERROR: cannot launch node of type [pid/controller]: pid
ROS path [0]=/opt/ros/noetic/share/ros
ROS path [1]=/home/bsb/project11_ws/src
ROS path [2]=/opt/ros/noetic/share

### command_bridge_receiver_node

[ben/command_bridge_receiver-2] process has died [pid 1928, exit code 127, cmd /home/bsb/project11_ws/src/command_bridge/nodes/command_bridge_receiver_node.py __name:=command_bridge_receiver __log:=/home/bsb/.ros/log/97c771f4-da9a-11eb-973f-0242ac110002/ben-command_bridge_receiver-2.log].
log file: /home/bsb/.ros/log/97c771f4-da9a-11eb-973f-0242ac110002/ben-command_bridge_receiver-2*.log

### mission_manager

[ben/mission_manager-4] process has died [pid 1930, exit code 127, cmd /home/bsb/project11_ws/src/mission_manager/node/mission_manager_node.py __name:=mission_manager __log:=/home/bsb/.ros/log/97c771f4-da9a-11eb-973f-0242ac110002/ben-mission_manager-4.log].
log file: /home/bsb/.ros/log/97c771f4-da9a-11eb-973f-0242ac110002/ben-mission_manager-4*.log

### mbes_sim_node

```
[ben/mbes_sim-14] process has died [pid 1968, exit code 1, cmd /home/bsb/project11_ws/devel/lib/mbes_sim/mbes_sim_node.py /home/bsb/project11_ws/src/mbes_sim/data/US5NH02M.tiff __name:=mbes_sim __log:=/home/bsb/.ros/log/97c771f4-da9a-11eb-973f-0242ac110002/ben-mbes_sim-14.log].
log file: /home/bsb/.ros/log/97c771f4-da9a-11eb-973f-0242ac110002/ben-mbes_sim-14*.log
```

### command_bridge_sender_node

```
[ben/command_bridge_sender-16] process has died [pid 1974, exit code 127, cmd /home/bsb/project11_ws/src/command_bridge/nodes/command_bridge_sender_node.py __name:=command_bridge_sender __log:=/home/bsb/.ros/log/97c771f4-da9a-11eb-973f-0242ac110002/ben-command_bridge_sender-16.log].
log file: /home/bsb/.ros/log/97c771f4-da9a-11eb-973f-0242ac110002/ben-command_bridge_sender-16*.log
```

### joy_to_helm

```
[ben/joy_to_helm-19] process has died [pid 1987, exit code 127, cmd /home/bsb/project11_ws/src/joy_to_helm/nodes/joy_to_helm.py helm:=piloting_mode/manual/helm __name:=joy_to_helm __log:=/home/bsb/.ros/log/97c771f4-da9a-11eb-973f-0242ac110002/ben-joy_to_helm-19.log].
log file: /home/bsb/.ros/log/97c771f4-da9a-11eb-973f-0242ac110002/ben-joy_to_helm-19*.log
```

rolker, 2021-07-01:
pid: Seems like the pid node didn't get installed. You can install it manually (from memory: sudo apt install ros-noetic-pid?) but I'll look to see if I specified the proper dependency for rosdep to install it.

The rest are probably due to missing the python future module. Again, I should check to see if I can make rosdep help us with that one too.

bsb808, 2021-07-01:
Thank you @rolker 

* The ros-noetic-pid package did sort things out.
* For nodes such as https://github.com/rolker/command_bridge/blob/noetic/nodes/command_bridge_receiver_node.py  - am I correct that you are using Python 2?  (I'm new to using Python 3, so still sorting these things out.)   I thought that Noetic was exclusively Python 3.

rolker, 2021-07-01:
No, Python 2 is not used for the noetic branches. I do have the python-is-python3 package installed on my system to make python3 the default for #!/usr/bin/env python.

bsb808, 2021-07-01:
Thank you - the python-is-python3 package helped (I tend to be explicit with the shebang for python3).  I'm making progress.  Next step is to install gdal via apt packages, but I need to turn attention to other projects for a bit.  Thank you for the help.

bsb808, 2021-07-02:
I got it all running with docker!  Here are the instructions and Dockerfile if you are interested: https://bitbucket.org/gfoe/project12/wiki/Home

I've found working in Docker is a really good way to manage development environments for multiple projects.

I put in one PR to address a gdal issue with the latest apt repo version.

Thanks again for the help getting started.

closed 2021-07-02 by bsb808