rolker/unh_marine_autonomy #5: ben simulation doesn't work

faragon55, 2021-07-27:
@rolker @bsb808 @apob

Last week I pulled all the GFOE forked repos and the ben simulation no longer works. The error seems to be caused by not being able to launch ```sound_paly_node.py```. However, the simulation worked fine when I first pulled the repos. It seems that merging the GFOE and CCOMJHC branches after https://github.com/GFOE/project11/commit/eb7811b8242adb19f2364b78472a956965727c4c might have caused the issue. 

``` ERROR: cannot launch node of type [sound_play/soundplay_node.py]: sound_play ```

The very first time I built the code and everything worked I had pulled https://github.com/GFOE/project11/commit/31d0c10443230b02abf527ce4b55ccfb3a32e1b3

```
setting /run_id to 0adc898e-ee75-11eb-8e2c-fb51ddab20c9
process[rosout-1]: started with pid [14321]
started core service [/rosout]
process[ben/command_bridge_receiver-2]: started with pid [14328]
process[ben/mru_transform-3]: started with pid [14329]
process[ben/mission_manager-4]: started with pid [14330]
process[ben/dubins_curves-5]: started with pid [14331]
process[ben/path_follower/crab_angle_pid-6]: started with pid [14336]
process[ben/path_follower-7]: started with pid [14337]
process[ben/hover_action-8]: started with pid [14344]
process[ben/path_planner-9]: started with pid [14349]
process[ben/mpc-10]: started with pid [14357]
process[ben/helm_manager-11]: started with pid [14358]
Starting MPC node
process[ben/asv_helm-12]: started with pid [14360]
process[ben/asv_sim_node-13]: started with pid [14361]
process[ben/mbes_sim-14]: started with pid [14364]
process[ben/manda_coverage-15]: started with pid [14374]
Starting planner node
process[ben/command_bridge_sender-16]: started with pid [14379]
process[operator/remote_manager-17]: started with pid [14381]
process[ben/joy_node-18]: started with pid [14385]
process[ben/joy_to_helm-19]: started with pid [14392]
ERROR: cannot launch node of type [sound_play/soundplay_node.py]: sound_play
ROS path [0]=/opt/ros/noetic/share/ros
ROS path [1]=/home/fas/development/project11_gfoe/catkin_ws/src
ROS path [2]=/home/fas/project11/catkin_ws/src
ROS path [3]=/opt/ros/noetic/share
process[operator/camp-21]: started with pid [14395]
[ WARN] [1627347205.996981372]: Couldn't set gain on joystick force feedback: Bad file descriptor
Map cleared. Using empty map now.
QMetaObject::connectSlotsByName: No matching signal for on_maxSegmentLengthLineEdit_editingFinished()
QMetaObject::connectSlotsByName: Connecting slot on_alignmentComboBox_activated() with the first of the following compatible signals: ("activated(int)", "activated(QString)")
terminate called after throwing an instance of 'ros::InvalidNameException'
  what():  Character [$] at element [1] is not valid in Graph Resource Name [/$(arg robotNamespace/project11/heartbeat].  Valid characters are a-z, A-Z, 0-9, / and _.
[operator/camp-21] process has died [pid 14395, exit code -6, cmd /home/fas/development/project11_gfoe/catkin_ws/devel/lib/camp/CCOMAutonomousMissionPlanner /home/fas/development/project11_gfoe/catkin_ws/src/camp/workspace/ /home/fas/development/project11_gfoe/catkin_ws/src/camp/workspace/13283/13283_2.KAP __name:=camp __log:=/home/fas/.ros/log/0adc898e-ee75-11eb-8e2c-fb51ddab20c9/operator-camp-21.log].
log file: /home/fas/.ros/log/0adc898e-ee75-11eb-8e2c-fb51ddab20c9/operator-camp-21*.log
[INFO] [1627347208.368147, 0.000000]: State machine starting in initial state 'PAUSE' with userdata: 
    []
[ WARN] [1627347210.856078762, 1627347210.450721549]: Timeout waiting for udp_bridge remote advertise service
```

rolker, 2021-07-27:
Sorry about that!
I added the ability for Camp to send audio alerts for mode changes which uses sound_play.

Rerunning `rosdep install --from-paths src --ignore-src -r -y` in catkin_ws should pick up the new dependency.

If not, you can install it yourself with `sudo apt install ros-noetic-sound-play`.

Alternatively, you can comment out the sound play node from the appropriate launch file.

bsb808, 2021-08-03:
I believe there are two problems evidenced in this issue.

1. The added dependency issue that @rolker discusses.
2. Also the camp node launch command fails with the error: 
```
 what():  Character [$] at element [1] is not valid in Graph Resource Name [/$(arg robotNamespace/project11/heartbeat].  Valid characters are a-z, A-Z, 0-9, / and _.
[operator/camp-21] process has died [pid 14395, exit code -6, cmd /home/fas/development/project11_gfoe/catkin_ws/devel/lib/camp/CCOMAutonomousMissionPlanner /home/fas/development/project11_gfoe/catkin_ws/src/camp/workspace/ /home/fas/development/project11_gfoe/catkin_ws/src/camp/workspace/13283/13283_2.KAP __name:=camp __log:=/home/fas/.ros/log/0adc898e-ee75-11eb-8e2c-fb51ddab20c9/operator-camp-21.log].
```

I believe 2 is addressed by this commit: https://github.com/GFOE/project11/commit/dd6b840c342c62b69d0922bfb646c93fba807c55 
which is a part of this PR:
https://github.com/CCOMJHC/project11/pull/10

bsb808, 2021-08-04:
I believe this is addressed with the merge of https://github.com/CCOMJHC/project11/pull/10  
Fine to close on my end.